itemisCREATE/statecharts #108: Add Type Conversion for String and Number types

andreasmuelder, 2015-05-08:
It should be possible to write 
var x : string
var y : integer
var z : string = x + y
Like in Java

andreasmuelder, 2015-05-18:
Created branch String_Type_Conversion

bohl-itemis, 2015-05-20:
added testmodel (.sct) and testcase (.sctunit)
added generated tests for Java, interpreter, C and C++ (c and c++ still fails)

andreasmuelder, 2015-05-20:
@muehlbrandt can you extend C and C++ code generator?

muehlbrandt, 2015-05-20:
Should be possible by the end of this week.

andreasmuelder, 2015-07-13:
Anything new on this, Markus?

muehlbrandt, 2015-08-07:
The problem with string concat in C is the dynamic memory allocation which is necessary to increase the string buffer sizes. An advantage of the currently generated code base is that it is internally completly free of dynamic memory allocation.

The implementation of memory allocation often depends on the used target platform. In the context of embedded systems it is often not possible to dynamically allocate memory because of security requirements or due to limitations of the target platform. For this reasons i would avoid to do the dynamic memory allocation within the generated statemachine code.

A possible solution could be to do the string concatenation within some external functions which will be part of the required API. So the client has to implement the concat functionality and has full control of the memory management.

To make things easier we could provide a default implementation of the concat function with memory management which could be compiled with the GNU Compiler Collection.