itemisCREATE/statecharts #112: Statemachine - use setters for updating variables value

smadarsz, 2015-06-22:
Would be helpful if the code generators will generate a call to variables setters when generating code that update the value of variables.
Motivation: 
For the integrated execution of LSC and SC we implement a new code generator (PlayGoStatemachine) that adds a relevant code for informing PlayGo on events raised by statecharts and on value changes of variables (such that are not internal). For this, it would be nice if we could add the relevant code only in the variables setters, rather then locate all the places in which variables may be set with an updated value, and add our code in many locations.

terfloth, 2015-06-23:
Yes - that makes sense for all those cases where extended code generator want to handle variable changes. 

There are some cases that we have to consider:
* No setters are generated for readonly variables: we would have to add setters in the implementation classes.
* No accessors are generated for internal variables: as local variables are not visible from outside we could ignore them in the first step.

The general use of setters instead of direct variable assignments seem will be no issue for Java and in most cases also C++ code generators. Nevertheless in the case of the C code generator the situation is a little bit different. By default additional overhead (non required setters) should be avoided. Nevertheless there may also be cases where extended setters may be used in C. As we want to keep the code generators as comparable as possible we should consider:

* Make the use of setters an option for the code generators. 
* Enable the general use of setters for the Java code generator by default.
* Enable the general use of setters for the C++ code generator by default.
* Disable the general use of setters for the C code generator by default.

muehlbrandt, 2015-06-23:
The Java generator already supports an interface observer feature. If this feature is enabled an additional interface is created which clients could implement to register themselves as listener to the statemachine. The statemachine then notifies the listeners for outgoing events.

So currently this callback is only generated for outgoing events but it could be easily extended to support notifications for modifications of each interface element (e.g variables, local and incoming events).

andreasmuelder, 2015-08-28:
@smadarsz Does the method described work for you? Can this issue be closed?

smadarsz, 2015-09-01:
@andreasmuelder I rather have the setter solution implemented. 
At this point I don't use the observer since it does not cover all my needs, plus it is currently implemented only for outgoing events, so it requires additional work from your end. So if possible, the setter solution will serve better my needs. Thanks...

terfloth, 2015-10-16:
fixed with issue #134 . You can now overwrite setters and getters.

closed 2015-10-16 by terfloth