itemisCREATE/statecharts #120: Question: what is the difference between in event and out event?

smadarsz, 2015-07-16:
What is the difference between in event and out event?

I understand that in/out logically defines the event direction, but is there a technical difference in the way the statemachine refers to them and process them?

smadarsz, 2015-07-16:
I see now that statemachine publishes its out events to its registered listeners (given that the InterfaceObserverSupport feature is enabled).
Is there any other difference?

terfloth, 2015-07-16:
The accessor methods in the interface will be different. For a declaration:

interface: 
	in event i
	out event o

the generated interface will look like:

	public interface SCInterface {
		public void raiseI();
		public boolean isRaisedO();
	}

closed 2015-08-28 by andreasmuelder