itemisCREATE/statecharts #134: Use internal setters and getters for variable access

muehlbrandt, 2015-09-14:
To support better code wrapping for implementations which extends a statemachine the internal access (within expressions) to variables should be realized with getters and setters instead of a direct variable access. The visibility of local variable access methods should be protected to allow their overriding in subclasses.

muehlbrandt, 2015-09-17:
Changes of the generated Java code:
- To access the variables within expressions getter and setter methods are used now (instead of directly accessing the variables).

- The visibility of the interface implementation inner classes has been changed to "protected" to allow their subclassing.

- If assignments are used as a expression (e.g. y = (x+=1)) then a protected "[type] assignVarname([type] value)" method is generated within the statechart (for internal scope) or interface (for interface scope) which is called within these expressions. For y = (x+=1) ```setY((assignX(getX() + 1)));``` is generated.

Example (based on the generated code of the traffic light example):
```java
package org.yakindu.sct.examples.java.trafficlight;

import org.yakindu.sct.examples.trafficlight.cyclebased.trafficlightwaiting.TrafficLightWaitingStatemachine;

public class CustomStatemachine extends TrafficLightWaitingStatemachine {
	
	//Overriding interface
	protected class CustomSCITrafficLight extends SCITrafficLightImpl {
		@Override
		public boolean getGreen() {
			//Add custom code
			return super.getGreen();
		}
	}
	
	protected class CustomSCIPedestrian extends SCIPedestrianImpl {
		@Override
		public boolean getRed() {
			//Add custom code
			return super.getRed();
		}
	}
	
	//Create and set new instances of the custom interfaces
	public CustomStatemachine() {
		super();
		sCITrafficLight = new CustomSCITrafficLight();
		sCIPedestrian = new CustomSCIPedestrian();
	}
}
```

terfloth, 2015-10-16:
Forked issue #143  for C++ generator.

closed 2015-10-16 by terfloth