itemisCREATE/statecharts #138: Text not showing in definition block

terfloth, 2015-10-10:
State names and declaration text is only visible during editing and disappears when the embedded editor is exited. See forum post: 

https://groups.google.com/forum/embed/?place=forum/yakindu-user#!topic/yakindu-user/G9F4fOAVTqE 

[image]
[image]
[image]

andreasmuelder, 2015-10-19:
@terfloth could you check this I do not have a Mac ;)

DominikTesch, 2015-11-11:
Hi
I can't reproduce your problem. I tried it on OSX El Captian with JDK 1.8 and 1.6 and Elcipse Luna.

closed 2016-03-30 by terfloth