itemisCREATE/statecharts #139: Access the name value of states

arianaa30, 2015-10-12:
Currently it is not possible for Yakindu Statecharts to access the name value of states, regions, or other environmental values.

terfloth, 2015-11-04:
What is the exact background of this feature request? As there are different possible approaches a short description of the usage scenario/use case would be great.

arianaa30, 2015-11-04:
One thing was that we needed to know when a machine is now in a specific state. So a function like State1.isNowActive() which returns true or false given if the state is active. We can do that via flags inside states, but we should unset them whenever we are in other states- too much work!

andreasmuelder, 2015-11-06:
There is a build in function active() (Press CTRL+space within the active function to get a list of all available states) Does this solve the issue?

arianaa30, 2015-11-06:
I see. So active() basically returns list of all "active" states, is that correct? And what is the return type?

andreasmuelder, 2015-11-06:
active takes the name of the state as an argument, for example
for example  active(main_statechart..main_region.A)  returns true if state A is active, false otherwise.

arianaa30, 2015-11-06:
That would be best!

andreasmuelder, 2015-11-06:
This is already implemented!
#Fixed

closed 2015-11-06 by andreasmuelder