itemisCREATE/statecharts #140: Raise event based on operation callback

arianaa30, 2015-10-12:
Currently we can’t directly raise events on the state machine from the operation callback. I need to receive a value over socket network in my operation, and raise an event based on that. This is not yet supported.

aboseley, 2015-10-14:
I think this is by design.  There are no event queues in yakindu you need to queue events outside the statemachine if this is what you need.  

https://groups.google.com/forum/embed/?place=forum/yakindu-user#!topic/yakindu-user/O9W7XJeIpcU

terfloth, 2015-11-04:
This feature request is qualified as relevant. As this is an requirement that correlates with the execution of multiple state machines and Java classes within a simulation it will likely be part of a larger topic.

BeckmaR, 2019-05-24:
This is now possible because in event queues have become available.

closed 2019-05-24 by BeckmaR