itemisCREATE/statecharts #144: Shapes disarranged after moving project

Prokli, 2015-10-18:
I tried to copy an entire Yakindu project from one to another PC (just a simple copy & past). 
When I open this project on the second PC all shapes and transition pointers are complete disarranged. 
Both PCs are connected to monitors keeping a screen resolution of 1920 x 1080. I copied all files of my project on PC 1.
OS on PC1 ist Windows 7 (Prof) and OS on PC2 is Win 8.1

terfloth, 2016-01-06:
Take a closer look at symptoms. Problems are also known across different target OSs

andreasmuelder, 2018-09-18:
This is related to different font scaling settings on windows. Fixed with the latest release for windows, still an issue with mixed platforms.

Closing as duplicate of https://github.com/Yakindu/statecharts/issues/735

closed 2018-09-18 by andreasmuelder