itemisCREATE/statecharts #153: Transition arrow routing is insane!!

Timmmm, 2015-10-23:
Trying to get nice neat transition arrows is an excercise in frustration. Whenever you move one node it moves all the others. I assume this was intended to be helpful - and very occasionally it is - but 99% of the time it just messes everything up.

It would be a massive improvement if moving a transition node *only moved that node*.

andreasmuelder, 2015-10-23:
Do you mean the behavior mentioned here?
http://muelder.blogspot.de/2013/11/5-usability-flaws-in-gmf-and-how-to-fix.html
1. Remove dithering connection anchors

Timmmm, 2015-10-23:
Yeah that's part of it. Also see a bit of this video: https://youtu.be/ev5wEjvje78?t=348

Notice how the middle nodes jump around randomly even when they aren't being directly manipulated. Another issue is if you drag and drop the end of a transition even slightly outside a state it jumps back to its original position rather than snapping to the nearest state. With composite states you only have a tiny hit target around the edge.

I can make a video demonstration if you like?

andreasmuelder, 2015-10-26:
A short video demonstrating the main pain points would indeed be helpful!

Timmmm, 2015-10-26:
Here are a few:

[If you move the ends of a transition it moves all the other nodes](http://gfycat.com/EnviousInexperiencedColt). I would have expected it to either a) Only move the end node, or b) Move the end node, and maybe the node that is connected to it in order to maintain orthogonal lines. It does neither and basically always messes up the entire transition line.

[Resizing a state moves the ends of the transitions](http://gfycat.com/DeadlyHauntingFinch) even when it doesn't have to. Also because it moves some of the ends, that moves the intermediate nodes of those transitions which messes them up (as a result of the previous issue).

A similar thing happens [when you move states](gfycat.com/FragrantMadeupBrownbutterfly). All the attached transitions get messed up.

Finally [the target for moving the ends of arrows for composite states is very small](http://gfycat.com/HoarsePeacefulBass). It looks kind of contrived in the video but this honestly happened to me a lot before I realised that I was dropping the arrows *just* outside the valid target. The solution would be to make the border a bit thicker (not too much or it will look stupid), and to make transition ends snap to states when they're just outside them.

Hope this helps!

andreasmuelder, 2015-10-26:
Thanks very much for the detailed bug report!

andreasmuelder, 2016-11-29:
@SimonEit please provide an expert review on rectangular routing

SimonEit, 2016-11-30:
I'm on it.

SimonEit, 2017-01-12:
@Timmmm Even though some time has passed, were still thankful for your feedback and use it as basis for improvement. Most of your reported behaviour is the result of manually adjusted transitions that are still in the _oblique_ setting (in the properties view).

You should notice a huge improvement during layouting of your model transitions if you change the default routing style to _rectilinear_ via the preferences ( > YAKINDU SCT > Diagram Appearance Panel). 

Thank you alot for your detailed feedback.

closed 2017-01-20 by andreasmuelder