itemisCREATE/statecharts #156: Comment

arianaa30, 2015-10-26:
Currently Yakindu does not allow putting comments in states, transitions, or variable declaration section. We have a large model, and it could be great if you could allow putting comments. (e.g. // similar to Java syntax)

andreasmuelder, 2015-10-26:
Comments work for me as expected in the textual definitions with a //.
what is the error you get when using this comment syntax?

arianaa30, 2015-10-26:
Oh, interesting! Resolved! There were nothing mentioned in the documentation and we thought there is no such feature!

andreasmuelder, 2015-10-26:
Sorry, the documentation is a little bit outdated. In case you want to update it:
https://github.com/Yakindu/statecharts/tree/master/plugins/org.yakindu.sct.doc.user/help

;-)

In the upcoming version there is even a comment integration for TODO and FIXME comments that are included in the Eclipse Task View.

closed 2015-10-26 by andreasmuelder