itemisCREATE/statecharts #162: Execute generator from command line

danlapid, 2015-11-04:
Being able to generate src files from .sct and .sgen from the command line, mostly used for build systems.

HunterHudson, 2015-11-04:
+1 - this would be very useful. I'd much rather only check in the diagram files and have the generated code created at build time. This would keep the diagram in sync with the source. This avoids the case when someone tries to directly modify the generated code.

tstellanova, 2016-01-12:
+1  --  need to be able to generate code from state files using standard build tools.

taylor-peterson, 2016-01-16:
+1 - this is essential for adoption

klausuhl-intel, 2016-01-27:
+1

* The command line generator should only work on the .sct file and read the configuration that is currently in the .sgen file from the command line. It may have to use additional files that are referenced via command line parameters to read, e.g., the license text.
* The command line generator has to be able to generate to any folder in the file system (esp. outside the source tree). It must not rely on an Eclipse project to be available.
* The command line generator should ideally be implemented as a plain Java program and not as a headless Eclipse product. Starting up a headless Eclipse instance is very time consuming. If you are having a large project with a lot of state chart models, the time that a headless Eclipse generator is wasting just for initialization will hurt you.

andreasmuelder, 2016-02-03:
There are basically 3 different approaches to execute the code generator headless.

1. Eclipse Application
As already mentioned, this is a very heavyweight approach because a whole headless eclipse has to be started. However, this would work right now because OSGI is running and an Eclipse workspace is available. A workspace build could easily be executed programmatically.

2. Plain Java Main
This is a very lightweight solution but it requires some refactorings because extension points are not available. Registration of Code Generators, Libraries and Domains must be done explicitly in java. Issue #358 is preliminary work for that. Furthermore  all dependencies to Eclipse File System must be replaced with java.io.file.

3. Plain Java with equinox
We could use equinox to load and initialize the extension points but execute the code generator without eclipse and EFS. This would be more flexible than a Plain Java implementation because new and custom code generators could be loaded without changing the code.

For version 1 + 2 at least the following command line parameters must be provided
* _filepath_ to the sgen model
* _folder_ to all statechart models referenced by the sgen model (since we can not use the xtext index here)
* _filepath to the workspace_. sgens targetProject parameter from the Outline feature is used as a simple folder relative to the workspace folder then.

stmork, 2016-02-03:
I have no problem to provide some essential parameters to a headless build. I would prefer a light weight solution - may be via a maven mojo, ant plugin or some sort. I agree, that a complex ecosystem like Eclipse, Xtext, etc need a lot of resources so never mind ;-)

szarnyasg, 2016-03-05:
@andreasmuelder does the 1st approach work in Yakindu 2.5? I tried to prototype a headless Eclipse application for generating the code:

* [the code generator application](https://github.com/szarnyasg/yakindu-codegen/blob/master/hu.bme.mit.yakindu.codegen/src/hu/bme/mit/yakindu/codegen/YakinduCodeGeneratorApplication.java)
* [the SCT/SGEN files](https://github.com/szarnyasg/yakindu-codegen/tree/master/hu.bme.mit.yakindu.codegen.callhandling/model) (I tested these in Yakindu SCT and its code generator worked fine.)

I was not yet able to generate the code though, as a [`No Statechart selected in genmodel` error](https://github.com/Yakindu/statecharts/blob/master/plugins/org.yakindu.sct.generator.core/src/org/yakindu/sct/generator/core/impl/AbstractSGraphModelGenerator.java#L125
) occured (it was not printed due to the lack of a workbench).

Should I load the SCT file separately? Or is there any way to get it resolved automatically?

andreasmuelder, 2016-03-06:
@szarnyasg I did not test your code yet but maybe the Xtext Linker can't resolve the cross reference to the statechart model because the statechart is not indexed properly. You could try to execute a Workspace build programmatically before generating the code:
ResourcesPlugin.getWorkspace().build(IncrementalProjectBuilder.FULL_BUILD,null);
Let me know if this works!

szarnyasg, 2016-03-06:
@andreasmuelder thanks, I added that to the [code](https://github.com/szarnyasg/yakindu-codegen/blob/master/hu.bme.mit.yakindu.codegen/src/hu/bme/mit/yakindu/codegen/YakinduCodeGeneratorApplication.java#L97).

I think it initiated the build, but it throws the following exception: `Contains: Errors running builder 'YAKINDU SCT Builder' on project 'hu.bme.mit.yakindu.codegen.callhandling'.
java.lang.IllegalArgumentException: No default domain found!`
([full stacktrace](https://gist.github.com/szarnyasg/d1eb3bb5dfcb51ef72c5))https://github.com/szarnyasg/yakindu-codegen/blob/master/hu.bme.mit.yakindu.codegen/src/hu/bme/mit/yakindu/codegen/YakinduCodeGeneratorApplication.java#L97.

I added the `org.yakindu.sct.domain` dependency, so now I have access to the `DomainRegistry` class. The `DomainRegistry.getDomainDescriptors()` method returns an empty list. Do I have to register the Java code generator here?

andreasmuelder, 2016-03-06:
@szarnyasg try adding a dependency to org.yakindu.sct.domain.generic

szarnyasg, 2016-03-06:
@andreasmuelder thanks! I'm almost there. After some fiddling with the Apache Batik dependencies (the launch config required both Batik 1.6 and 1.7 to be present), it worked fine. The output is now a lot longer: https://gist.github.com/szarnyasg/461b454e5378a40ab886, but I cannot find the generated java files (I searched the workspace, the project dir, the statechart project dir, etc.).

szarnyasg, 2016-03-14:
I managed to solve it, using the code from https://github.com/Yakindu/statecharts/blob/master/test-plugins/org.yakindu.sct.generator.c.test/src/org/yakindu/sct/generator/c/gtest/GTestHelper.java. The solution is in https://github.com/szarnyasg/yakindu-codegen with a README file.

It turned out that I had to load both the sct and the sgen models: https://github.com/szarnyasg/yakindu-codegen/tree/master/hu.bme.mit.yakindu.codegen/src/hu/bme/mit/yakindu/codegen

terfloth, 2016-03-30:
@jdicks : add How-To documentation for this feature

jdicks, 2016-04-12:
an example how standalone execution can be done could be found [here](https://github.com/Yakindu/statecharts/tree/master/examples/org.yakindu.sct.standalone).

closed 2016-04-12 by jdicks

JackUkleja, 2016-11-01:
This is also an essential feature for us. We use TeamCity for our mostly C#/C++ product and like to ensure everything checked in can be automatically verified as much as a possible. So we would want to ensure checked in statemachines are generated into source code, and that source code compiles and passes the requisite tests. Hope this feature is available soon and works easily with TeamCity on Windows. Cheers

elgeekomm, 2016-11-01:
+1 this is an essential feature for us. We use CMake for our build system and while Eclipse can spawn it, it doesn't correctly sequence Yakindu Statechart code generation with CMake. Once this feature is released we can do full continuous integration.

@JackUkleja this issue is closed, see https://github.com/Yakindu/statecharts/issues/675 which indicates this is part of the recent 2.8 release. I'm still looking to confirm this is the case.

andreasmuelder, 2016-11-28:
@danlapid @HunterHudson @stmork @tstellanova @taylor-peterson @klausuhl-intel @szarnyasg @JackUkleja @elgeekomm 

We just released the headless code generator, check out 
https://www.itemis.com/en/yakindu/statechart-tools/documentation/user-guide/#generating-state-machine-code-headless
for more information. Feedback welcome!

stmork, 2016-11-28:
Nice. Is it correct, that this feature is only available in Pro version?

jdicks, 2016-11-28:
hi steffen,
no this is not correct. the headless feature is free to use, but its not open source and just available within the pro version out-of-the-box. for the open edition you can install it from pro updatesite by using the 'Install New Software' wizard. 
regards!

stmork, 2016-11-28:
Thank you for your quick response. I cannot find the headless component at your update site but all other 2.9.x components. Is it already released, yet?

jdicks, 2016-11-28:
Hi Steffen, 
did you try the pro updatesite? On the pro updatesite you will find version numbers like 0.9.4 - see https://www.itemis.com/en/yakindu/statechart-tools/download-options-pro-edition/.

Regards!

jdicks, 2016-11-28:
Pre checked if it is available and it is:
[image]

stmork, 2016-11-28:
Oops! My mistake. I got it right. The first startup forced #1113. I will test it in the near future.

elgeekomm, 2016-11-28:
On Ubuntu 14.04, noticed that the executable /opt/eclipse/scc does not have executable privileges. After adding executable by "chmod +x scc" and running the script, I receive the error message:

"./scc: line 7: /opt/eclipse/SCT: No such file or directory"

andreasmuelder, 2016-11-28:
@elgeekomm thanks for your feedback I opened #1114 for executable privileges.
Is the reason for the error msg the same as mentioned in #1113 ?

stmork, 2016-11-28:
No, it's an additional problem.

elgeekomm, 2016-11-28:
@andreasmuelder following the advice in #1113, in the script scc, replacing the line

``$SCRIPTPATH/SCT --launcher.ini $SCRIPTPATH/headless.ini "$@"``

with

``$SCRIPTPATH/eclipse --launcher.ini $SCRIPTPATH/headless.ini "$@"``

appears to work correctly. I haven't tried code generation but I do see the help output.

FYI, using Oracle Java I receive a warning regarding a deprecated setting in headless.ini:

``Java HotSpot(TM) 64-Bit Server VM warning: ignoring option PermSize=256m; support was removed in 8.0``

elgeekomm, 2016-11-29:
I've integrated the headless code generation with CMake and it works beautifully. Thank you!

A few issues:
1. Regardless of what I pass as --baseDir, the output says "using base dir /"
2. When installing this feature as a plugin and not the standalone installer, the command SCT is not present in the system. The command I posted above resolves this issue.
3. In headless.ini the option PermSize=256m is deprecated for Oracle Java 8 and a warning is generated.
4. If a folder contains two .sct files and a single .sgen file that generates both of them, neither will be generated. Here's how to recreate:

src/my_target/diagram1.sct
src/my_target/diagram2.sct
src/my_target/my_target.sgen

The sgen file has two "statechart TaskStatechartDiagram" entries. In Eclipse, both are correctly generated. If I run the command:

``$SCRIPTPATH/eclipse --launcher.ini $SCRIPTPATH/headless.ini --baseDir src/my_target --model diagram1.sct --sgen my_target.sgen``

the output shows first it found 1/1 and then reports a reference error and 0/1 is generated. Let me know if you can't recreate and I'll try to capture the output. 

Thanks for all your hard work on this.

andreasmuelder, 2016-11-29:
Locking this conversation, if you find any additional bugs please file a seperate issue.

jdicks, 2016-11-29:
@elgeekomm we found an issue regarding the base directory and will provide a bugfix version soon. thanks for the fast and helpful feedback!