itemisCREATE/statecharts #168: Add a python code generator

Jean--Michel, 2015-11-13:
It would be great to generate Python code.

andreasmuelder, 2015-11-16:
here is a short introduction on how to write custom code generators:
http://statecharts.org/documentation.html#CreateCustomCodeGenerators
The best way to start is to check out our source code via Oomph as descriped here:
https://github.com/Yakindu/statecharts/blob/master/README.md
Then you should have a closer look at the plugin org.yakindu.sct.generator.java

We have a powerful generator testing framework called SCTUnit with a lot of test models so I can contribute a bunch of test cases using the Java Python Interpreter that ensure that the execution semantics work as expected.

terfloth, 2016-03-30:
@Jean--Michel : are there any news on the python generator?

MickDre, 2016-03-30:
Hi,

I am working with @Jean--Michel on this python generator.

We are currently trying different approach to succeed on this projet:
_ From the Java generator by adapting Java generated code to python code (In progress but nearly discarded due to its complexity)
_ From scratch (not started)
_ From the C++ generator by adapting C++ generated code to python code (In progress)
_ From a statemachine python library from the internet (not started)
_ From a statemachine python which we adapt step by step the generic parts (In progress)

I am currently working on the last approach. I do it as follow:
1) Create a java project with a lamp which can switch on/off or flash.
2) Generate its java state machine
3) Translate its state machine to python using j2py (java to python3 converter)
4) Correct the translated state machine
5) Add genericity with the sct graphs parsing (as you do in Java for example)
I am currently on the 4th step (TimerTask/Timer/lock etc are a bit hard to translate).
As soon as this step is done, and I have a fixed state machine for the lamp project in python, I will adapt the generator to parse the sct graph and build graph specific state machine.

As first experience feedback, I could say that a complete tutorial may be interesting to start from scratch. Same goes with example, which are, to me, not complete enough.

As second experience feedback, we also have some troubles with one developper to install the whole package to work too (depends on cases, I used the Oomph installer and didn't have troubles).
Did the procedure changes recently ?

If you have any question, we are willing to answer/work together on this subject.

Regards,
Mickael

andreasmuelder, 2017-11-21:
The Python code Generator was released today as a YAKINDU labs project:
https://blogs.itemis.com/en/python-code-generation-with-yakindu-statechart-tools

closed 2017-11-21 by andreasmuelder