itemisCREATE/statecharts #169: Timer does't work with ms, us and ns

DominikTesch, 2015-11-13:
[image]
If you use 'ms' on the left interface window your timer won't work in your Statechart simulation. But if you use 's' it will work. Moreover if you use 'us' and 'ns' you will get this error.
[image]

DominikTesch, 2015-11-13:
[image]
The .sct project as .jpg

andreasmuelder, 2015-12-18:
Fixed for M22

closed 2015-12-18 by andreasmuelder