itemisCREATE/statecharts #190: Code generators should support generation of model comments.

nyssen, 2015-11-25:
The code generators should transfer model comments into code. Especially those comments that are added to transitions, entry and exit actions, states, interfaces, and regions should be reflected in the source code.

terfloth, 2016-02-22:
First it has to be defined where the comments and the documentation should go.

robfox1976, 2016-11-10:
There are already generated comments in the generated code, for example;

```
/* Entry action for state 'State_B'. */
static void testing_enact_main_region_State_B(Testing* handle)
{
	/* Entry action for state 'State_B'. */
	handle->ifaceUser.currentState = 2;
}
```
In my example state model (see below) I have put comments in both states.
[image]
Could the example code generated become;

```
/* Entry action for state 'State_B'. */
static void testing_enact_main_region_State_B(Testing* handle)
{
	/* Entry action for state 'State_B'. */
	handle->ifaceUser.currentState = 2; /* set state A=2 */
}
```