itemisCREATE/statecharts #194: The (optional) name scrambling used by c code generator falls back to hashing too fast.

nyssen, 2015-11-25:
When specifying the maxIdentifierLength within the IdentifierSettings feature of the c code generator, the name scrambler falls back to generating hashed names too soon. E.g., when specifying a max identifier length of 8 chars, and having a simple statemachine with two states and a single transition (with a single input event), hashing is already performed. I would propose to use truncating as a de-escalation step first and to fall-back to hashing only as the last resort.

nyssen, 2015-12-02:
It might already bring a significant benefit if adjusting the name computation within DefaultNamingService.addShortVertexNames(). In case simple names are unique, no need to add a region (or even composite state) prefix.

muehlbrandt, 2015-12-02:
This behavior is already implemented. The short name strategy first uses the element name concatenated with the statemachine name prefix (defined within the genmodel) and checks if this name matches the maximal identifier length andif it is unique.

The statemachine name prefix is important if several statemachines are generated and linked together in case of common name intersections between statecharts. Currently this prefix must be set.

We can change this setting to be optional to save the prefix characters. In this case the developer has to care about name intersections between statemachines.

nyssen, 2015-12-02:
I think the prefix is ok (and it can already be removed - if desired - using the IdentifierSettings feature. I am referring to the case where the simple name is not unique. In this case, adding the region prefix does not seem to the best strategy. I would propose to fall back to the composite state that encloses the state (as a suffix), thus apply a 'postfix' notation, and to add a region index to this (again as a suffix) in case the state is not unique within the context of the composite state. That would deliver something like SM1_State1_Composite1_0_ and SM1_State1_Composite1_1 for two identically named states in different regions of a composite state in a statemachine SM1. The same could be 'recursively' applied for the composite state name, in case its not unique. 

After this procedure has been applied, we could still apply the scrambling, replacing as much suffix of the resulting identifier name as possible, and preserving as much prefix as possible.

paphko, 2017-03-28:
What is the current status here? MISRA rule 5.1 requires a max length of 31 and with that value, almost all identifiers are hashed.

By the way, I tried to set `maxIdentifierLength = 400` and I ran into an IllegalArgumentException:
```
java.lang.IllegalArgumentException: Unhandled parameter types: [null, REMOVE_VOWELS]
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.elementName(DefaultNamingService.java:889)
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.createShortName(DefaultNamingService.java:514)
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.createShortName(DefaultNamingService.java:565)
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.createShortName(DefaultNamingService.java:581)
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.getShortName(DefaultNamingService.java:489)
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.putShortName(DefaultNamingService.java:330)
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.addShortFctNames(DefaultNamingService.java:277)
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.createShortNameMap(DefaultNamingService.java:256)
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.initializeNamingService(DefaultNamingService.java:144)
	at org.yakindu.sct.generator.c.CNamingService.initializeNamingService(CNamingService.java:81)
	at org.yakindu.sct.generator.c.StatemachineHeader.content(StatemachineHeader.java:75)
	at org.yakindu.sct.generator.c.CGenerator.generateArtifacts(CGenerator.java:89)
	at org.yakindu.sct.generator.c.CGenerator.generate(CGenerator.java:80)
```

paphko, 2017-03-28:
I also get a name clash with max length 31 as shown in this screenshot:
[image]

BeckmaR, 2017-03-30:
Hey paphko,
thank you for contributing to the issue. As you can see above, I already put some work into this. I developed a name service that sorts all the names in a tree and shortens the names based on a heuristic, e.g. vowels and small letters are cut before the other ones and common parts are more likely to be cut than the individual parts (basically entropy based). For example the two names
* `main_region_StateA`
* `main_region_StateB`
would have the part `main_region` in common so that part would be cut earlier. That could result in:
* `mn_rgn_StateA`
* `mn_rgn_StateA`
or even
* `m_SttA`
* `m_SttB`
I personally believe that the names generated by this algorithm are still pretty okay to understand. What's your opinion on this?

The downside though is that this is still not built into SCT, because we would need to take the Naming Service apart completely and feared to break existing generated code. Because you are now the first one who ran into issues with this, we gave other issues a higher priority.

Name clashes are of course a real bug and we will probably increase the priority of this issue.

PR #1912 (2018-03-09): Naming Service Changes
This pull request changes the behavior of the naming service. It removes all name shortening functionality, which is instead moved to the C domain and implemented differently. 
By using the TreeNamingService from PR #949, it fixes #194 and fixes #195.
It fixes #194 and incorporates the TreeNamingService from PR #949. 

Includes also PR #1908, which can be reviewed on its own.

The changes made to the TreeNamingService (simplyfing, cleanup)  can be seen separately here: https://github.com/Yakindu/statecharts/compare/master...newNamingService
closed 2018-03-09 by terfloth