itemisCREATE/statecharts #207: Java Generated Statemachine stops working when using TimeEvents

DominikTesch, 2015-11-27:
https://groups.google.com/forum/embed/?place=forum/yakindu-user#!topic/yakindu-user/eXhggfFduM4
[image]

Java code:

package org.yakindu.scr.defaultsm;

import org.yakindu.scr.RuntimeService;
import org.yakindu.scr.TimerService;

public class Test1 {

	int temp;
	DefaultSMStatemachine sm = new DefaultSMStatemachine();

	public Test1() {
		sm.setTimer(new TimerService());
		sm.getSCInterface().getListeners().add(() -> System.out.println(++temp));
		sm.init();
		sm.enter();
		RuntimeService.getInstance().registerStatemachine(sm, 50);
	}

	public static void main(String[] args) throws InterruptedException {
		Test1 t1 = new Test1();
		while (true) {

			Thread.sleep(1000);
		}
	}
}

andreasmuelder, 2015-11-27:
The problem here is that a time event may arise after the time event is checked but before the time events are cleared. A potential solution may be a TimeEvent Buffer:

private final boolean[] timeEvents = new boolean[1];
private final boolean[] bufferedTimeEvents = new boolean[1];

at the beginning of runcycle:
synchronized (bufferedTimeEvents) {
           System.arraycopy(bufferedTimeEvents, 0, timeEvents, 0, 1);
           clearBufferedEvents();
 }

 public void timeElapsed(int eventID) {
        synchronized (bufferedTimeEvents) {
            bufferedTimeEvents[eventID] = true;
        }
    }

private void clearBufferedEvents() {
        for (int i = 0; i < bufferedTimeEvents.length; i++) {
            bufferedTimeEvents[i] = false;
        }
        
    }
}

muehlbrandt, 2015-11-27:
This problem also occurs in context with standard events. If a event is rised after it is checked by the statemachine but before the event is cleared it is lost. This behavior could happen if cycle execution and event raising is executed by concurrent threads.

hans1234567, 2015-11-27:
Hi,
thanks for the quick answer. The solution works, but this created another problem.
Internally an operation is called every 1s to display the current time (similar to the tutorial video). Now that the TimerService is synchronised the time will sometimes be displayed incorrectly, if other timed events occur.

andreasmuelder, 2015-12-14:
@muehlbrandt I move this to M23

PR #402 (2016-02-15): Issue 207 cyclebased statemachine wrapper
- Added wrapper for cycle based synchronized execution (fixes #207).

- Refactoring of generated RunnableWrapper.

- Added generation of observer support and operation callbacks for wrapper classes (fixes #308)
closed 2016-02-15 by muehlbrandt