itemisCREATE/statecharts #216: Model drawing issues

RainerKlute, 2015-12-02:
The graphical model representation has a couple of issues, exemplified in the attached PNG file with red indicator numbers: 
* Transition does not properly vertically align with the initial state (1).
* There's a gap between the initial state and the transition (2).
* The arrow should end at the outer edge of the state's border line or alternatively end in the middle of the border line. In any case, it should not extend into the state's box (3).
* The state box's blue color gradient extends too far to the right and appears outside the box (4).
* Lines should either be rounded, mitered or skewed at the corners. In other words, corners should look nice (5).
* Box shadows should be smooth (6).
* State names are not centered, extend too far to the right and are clipped (7).  
 
[image]

andreasmuelder, 2018-09-18:
@tangele can you have a look which of these issues are already fixed in the current version?

tangele, 2018-10-04:
still TODO:

* Transition does not properly vertically align with the initial state (1). 
* Box shadows should be smooth (6).

everything else is fixed in current [master ](https://github.com/Yakindu/statecharts/commit/f607ce8f13806032adf60c99a488758e449668d6)

BeckmaR, 2018-10-08:
We shouldn't fix the shadows, me thinks...

andreasmuelder, 2018-11-06:
Fixed in current master

closed 2018-11-06 by andreasmuelder