itemisCREATE/statecharts #222: User doc chapter 'Installation' does not open in Eclipse help

terfloth, 2015-12-04:
The 'Installation' chapter cannot be opened in the Eclipse help due to an error (see in comment). The causer seems to be a missing closing dic tag. 

Do the following:
1. define a text case (manual or automated) to check issues of this kind.
2. fix the problem

terfloth, 2015-12-04:
An error occured while processing the requested document:

org.xml.sax.SAXParseException; lineNumber: 245; columnNumber: 40; Elementtyp "div" muss mit dem entsprechenden Endtag "</div>" beendet werden.
	at com.sun.org.apache.xerces.internal.parsers.DOMParser.parse(DOMParser.java:257)
	at com.sun.org.apache.xerces.internal.jaxp.DocumentBuilderImpl.parse(DocumentBuilderImpl.java:348)
	at org.eclipse.help.internal.dynamic.DocumentReader.read(DocumentReader.java:56)
	at org.eclipse.help.internal.dynamic.XMLProcessor.process(XMLProcessor.java:49)
	at org.eclipse.help.internal.xhtml.DynamicXHTMLProcessor.process(DynamicXHTMLProcessor.java:66)
	at org.eclipse.help.internal.webapp.servlet.DynamicXHTMLFilter$1.close(DynamicXHTMLFilter.java:83)
	at java.io.FilterOutputStream.close(FilterOutputStream.java:159)
	at org.eclipse.help.internal.webapp.servlet.FilterHTMLHeadAndBodyOutputStream.close(FilterHTMLHeadAndBodyOutputStream.java:314)
	at org.eclipse.help.internal.webapp.servlet.PluginsRootResolvingStream.close(PluginsRootResolvingStream.java:222)
	at java.io.FilterOutputStream.close(FilterOutputStream.java:159)
	at org.eclipse.help.internal.webapp.servlet.FilterHTMLHeadOutputStream.close(FilterHTMLHeadOutputStream.java:162)
	at org.eclipse.help.internal.webapp.servlet.EclipseConnector.transfer(EclipseConnector.java:203)
	at org.eclipse.help.internal.webapp.servlet.ContentServlet.doGet(ContentServlet.java:47)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:687)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.equinox.http.registry.internal.ServletManager$ServletWrapper.service(ServletManager.java:180)
	at org.eclipse.equinox.http.servlet.internal.HttpServiceRuntimeImpl$LegacyServlet.service(HttpServiceRuntimeImpl.java:1232)
	at org.eclipse.equinox.http.servlet.internal.registration.EndpointRegistration.service(EndpointRegistration.java:153)
	at org.eclipse.equinox.http.servlet.internal.servlet.ResponseStateHandler.processRequest(ResponseStateHandler.java:63)
	at org.eclipse.equinox.http.servlet.internal.context.DispatchTargets.doDispatch(DispatchTargets.java:98)
	at org.eclipse.equinox.http.servlet.internal.HttpServiceRuntimeImpl.doDispatch(HttpServiceRuntimeImpl.java:372)
	at org.eclipse.equinox.http.servlet.internal.servlet.ProxyServlet.service(ProxyServlet.java:70)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.equinox.http.jetty.internal.HttpServerManager$InternalHttpServiceServlet.service(HttpServerManager.java:357)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:808)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:587)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:221)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1127)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:515)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1061)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:97)
	at org.eclipse.jetty.server.Server.handle(Server.java:499)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:310)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:257)
	at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:635)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
	at java.lang.Thread.run(Thread.java:745)

andreasmuelder, 2015-12-07:
Something that Rainer should fix

closed 2015-12-08 by terfloth