itemisCREATE/statecharts #257: Special treatment for sc_types.h is not documented.

nyssen, 2015-12-14:
When generating source code, the sc_types.h is specially treated in a sense, that it is not overwritten and manual adoptions are preserved (in contrast to all other files generated by the c code generator). This special treatment is currently not documented. 

As we have now introduced general support for this kind of treatment for static files (libraryTargetFolder) we could also think of removing it instead of documenting it. IMHO that would be more consistent.

RainerKlute, 2016-02-16:
@nyssen, does this apply to the C code generator only or to the C++ code generator, too?

I am going to document this special treatment now, but I'll mark it as deprecated and point out that this (mis-)feature might be removed in the future.

nyssen, 2016-02-16:
This holds for both, C as well as C++.

RainerKlute, 2016-02-16:
Fixed with PR #356

closed 2016-02-16 by RainerKlute