itemisCREATE/statecharts #259: Get current state array

danlapid, 2015-12-14:
I'd like to be able to get an array of current states, maybe only in non-orthogonal state machines even.
I'd like the stateConfVector to be public or have get/set and I think that's a feature many people would have use for, having to make a lot of if/elses for a function that will then run through a big switch/case sounds like a bad thing to have in an embedded system..

andreasmuelder, 2015-12-27:
Maybe we can add a generator feature for a read only access to the current state configuration vector

terfloth, 2016-01-06:
@danlapid : What code generator do you use? For C you can access by handle->stateConfVector. 

We will add public readonly access for Java and C++.

wixiw, 2020-08-14:
Hi, 

+1 on this feature. We are currently writing a switch case from "isStateActive()" which is very weird.
We agree that introspection is bad for operational code, nevertheless we like to log every state transition of our high level machines.