itemisCREATE/statecharts #260: Client code missing in "trafficlight" example

RainerKlute, 2015-12-15:
The client code described in the reference documentation is missing. The documentation reads as follows:

"You can checkout the Java sample project _"org.yakindusct.examples.trafficlight":https://github.com/Yakindu/statecharts/tree/master/examples/org.yakindu.sct.examples.trafficlight_ from the "YAKINDU Statechart Tools GitHub repository":https://github.com/Yakindu/statecharts. The Java example contains statechart, SGen model, **graphical widgets, and some glue code to connect the generated code with the widgets.** The graphical widgets are based on SWT."

The components in bold are missing, at least in the project generated by the examples wizard. Consequently, the following fails:

"To execute the Java example, run the file _CrossingDemoCycleBased.java_ as "Java Application" from the Eclipse _Run As_ context menu."

andreasmuelder, 2015-12-15:
since our product bundle is not shipped with PDE anymore, The plugin dependency to SWT could not be resolved. We plan to build the example based on Swing or JavaFX for the next version.

RainerKlute, 2015-12-15:
Means the documentation sucks until then. Oh, well.

andreasmuelder, 2015-12-15:
Yeah. Go ahead and create a new example that we can bundle. JavaFX does not solve the dependency problems, so Swing is the technology of choice.

terfloth, 2015-12-15:
I close this issue as we have first should specify what examples we need for what purpose and how we make them accessible.  So far we will have to fix the documentation ( see #262 ) .

closed 2015-12-15 by terfloth