itemisCREATE/statecharts #263: Yakindu SC_Types problems C/C++

danlapid, 2015-12-16:
Hey so I was talking to another engineer on the team and he asked me to forward you the following concerns:
You shouldn't use define instead use typedef 
Const char* instead of char*

The type char * is unsafe to be used as a default string type because it indicates to the compiler that the contents of the string are allowed to be modified freely, but the type does not contain adequate intrinsic size information for this to be done safely.

It would be highly preferable that if the tooling is not modifying the values using sc_string, that they be const char * instead.  This will enable their use with a broader set of type safe code, as well as enabling proper interaction with string literals.

Finally, the use of a #define to create sc_string is not a safe modern way of using c/c++.  It is preferable that new types like this are consistently defined in the language via typedefs to mitigate collisions in the preprocessor.

nyssen, 2015-12-23:
I think that pretty much resembles the discussion at http://stackoverflow.com/questions/1674032/static-const-vs-define-vs-enum-in-c, where a third alternative (enum) seems to be favored. In contrast, standards like Autosar intensively make use of #define (as currently done in C generator).  

Maybe a generator option would be the best way to go here, so that users could decide what concrete alternative to use.

terfloth, 2016-06-01:
The current solution was introduced to fix issue #121 .

terfloth, 2017-02-27:
For now we will do not change the sc_string definition because:

- the current solution solves issue #1210 
- sc_types.h is intended to be used for manual customization in order to adapt the type definitions to specific needs.
- next steps regarding string handling will be done in issue #1233

closed 2017-02-27 by terfloth