itemisCREATE/statecharts #266: Add support for generating tracing callbacks (in C generator).

nyssen, 2015-12-23:
We should create a code generator feature that enables creation of callbacks, via which entering and exiting of states an be traced. The callbacks could be generated into the 'Required' header. It should pass a respective state identifier.

nyssen, 2015-12-23:
Merged enhancement into master. Resolving as fixed.

terfloth, 2016-01-04:
Reopened since the generated code is not compilable and complete.

In detail there are the following problems:
1. The required header declares the callback function prototypes "extern void stateEntered(const SCTStates state);" and "extern void stateEntered(const SCTStates state);" wich uses the undefined type SCTStates
2. The callback function names are not scoped by a prefix as other callback functions but should be.
3. It is not possible to distinguish different state machine instances in the callback implementation as the handle is not provided as parameter.

terfloth, 2016-01-04:
We should define test cases for this feature. At least we require an entry for the manual test plan.

nyssen, 2016-01-06:
I will take a look into it.

nyssen, 2016-01-06:
I fixed the issues you named. What remains is to define test cases for this functionality.

closed 2016-04-04 by terfloth