itemisCREATE/statecharts #271: Variables from operations are not accessible in C++ generated code.

stmork, 2015-12-26:
If I code the following:

internal:
    var muted :  boolean
    operation Mute()

The muted variable is not accessible from the Mute() method. This is true for internal scope and from any interface scope when using C++ generated code.

andreasmuelder, 2015-12-27:
afaik this is by design - it is not allowed to read or write internal variables outside of the statemachine.
if the variable is contained in an interface it should be possible to access the variable via a getter.

stmork, 2015-12-27:
> afaik this is by design - it is not allowed to read or write internal
> variables outside of the statemachine.
> if the variable is contained in an interface it should be possible to
> access the variable via a getter.
It makes sense for internal operations to access internal variables. I
don't want to access the variables from outside. Even from an external
visable interface it was not possible to access the variables out of the
operations from the same scope.

I used the workaround to set the innerFunctionVisibility = "protected"
generator property.

closed 2015-12-27 by andreasmuelder