itemisCREATE/statecharts #273: No documentation about headless build available

stmork, 2015-12-26:
I would prefer to build the statemachines artifacts with Maven headlessly. Is there any example of a pom.xml out there?

andreasmuelder, 2015-12-27:
Wouls be very useful, but it requires some refactoring to remove dependencies to the eclipse file system to allow headless execution.
Duplicate of #162

closed 2015-12-27 by andreasmuelder

stmork, 2015-12-27:
> Wouls be very useful, but it requires some refactoring to remove
> dependencies to the eclipse file system to allow headless execution.
> Duplicate of #162 <https://github.com/Yakindu/statecharts/issues/162>
Is there no pom.xml or something like this outside? This forces me to
check in the generated artefacts into version control which is not best
practice.

andreasmuelder, 2015-12-27:
This is not yet implemented but contributions are always welcome :)

stmork, 2015-12-27:
> This is not yet implemented but contributions are always welcome :)
My job is to implement a statemachine for an embedded system. May be
that I would contribute something to SCT in future but my job is not to
fix your bugs and do your work.

Have a happy new year 2016

stmork, 2016-10-06:
I've tried the example @jdicks posted on your website (https://blogs.itemis.com/en/how-to-standalone-generate-statechart-code-for-continuous-integration-builds). Unfortunately it does not work with the C++ generator. The reason is that the package of the CPPDefaultFeatureValueProvider is not reexported in the manifest file.

Btw. Is it possible to define the feature libraries using platform URIs like platform:/plugin/...? This seems to be better than copying/extracting the XMIs from the git repository.