itemisCREATE/statecharts #276: No project save working while editing statechart description language

stmork, 2015-12-27:
If I make changes to the statechart description language it is not possible to save the statechart. As recently as I click back into the diagram saving works successfully.

svenjawendler, 2017-05-02:
The editor only becomes dirty when I leave the definition section.

SimonEit, 2017-08-22:
@svenjawendler just retested this issue. It is fixed. Please close.

SimonEit, 2017-09-07:
I just reproduced this issue in 3.0.1. It's still there. I can see how the users expectation would be that you can save you changes in the statechart description with ctrl-s at any point in time. This issue may even result in loss of changes, i.e. if the user makes changes in the description assumes ctrl-s saves it and then closes the statechart tab. No warning or autosave is displayed.

andreasmuelder, 2018-01-02:
This becomes obsolete with the new statechart specification section: #1442

closed 2018-01-02 by andreasmuelder