itemisCREATE/statecharts #277: Missing print preview/JPEG save

stmork, 2016-01-02:
For documentation purposes we have to print or save the statechart into file. I found no possibility how to do that.

andreasmuelder, 2016-01-03:
You can right click on the canvas and select Save as Image File

stmork, 2016-01-03:
No, I'm not able to save. I've attached a screen shot.
[image]

andreasmuelder, 2016-01-03:
I can't reproduce this on Windows 64 bit with JDK 8.
Which OS and JDK do you use? Does it work for BMP, PNG etc?

stmork, 2016-01-03:
Ubuntu 14.04.3 LTS and OpenJDK7. I need only JPEG. So please test yourself.

RainerKlute, 2016-01-06:
This functionality depends on the Java Runtime Environment: I was able to reproduce this bug on Linux (64bit) with OpenJDK; however, everything works fine with Oracle JDK.
Please change your JDK or export as PNG and convert to JPEG externally.

closed 2016-01-06 by RainerKlute