itemisCREATE/statecharts #280: Convert documentation to DocBook

RainerKlute, 2016-01-06:
Subtasks:
* convert source files
* adapt build process
* adjust HTML formatting
* adjust PDF rendering
* …

nyssen, 2016-01-07:
The currently used wiki-text supports generation of DocBook. Why switch the underlying format?

RainerKlute, 2016-01-07:
DocBook has a very rich content model which outpaces textile (wiki text) by far. Using DocBook XSL, the documentation source can be converted into a multitude of output formats, including Eclipse Help, HTML, PDF, and EPUB.

Some more goodies:
* Extensible output formatting/generation
* Headline enumeration
* Conditional text
* Index
* Glossary
* …

andreasmuelder, 2016-01-26:
@RainerKlute Did you close this because it is fixed? Is there a pull request fixing this issue?

RainerKlute, 2016-01-26:
It's closed, because it's fixed. However, it's in the userdoc_structure_2.0 branch only.

andreasmuelder, 2016-01-26:
Can you create a PR to merge this into master?

RainerKlute, 2016-01-26:
I think merging the userdoc_structure_2.0 branch into the master would not be sensible at this time, because that branch comprises the new documentation structure, which is still work in progress. We could consider to transfer select contents of the new documentation to the old textile format, but we might also consider that a waste of time – time that should better be spent on rewriting the documentation to fill the new structure appropriately.

RainerKlute, 2016-02-03:
Merged into master branch.

closed 2016-02-03 by RainerKlute