itemisCREATE/statecharts #287: set the value at startup and cannot be changed later

DominikTesch, 2016-01-08:
reported via forum

I encountered the same issue. (I'm using the latest Yakindu bundle for Windows, 64-bit).

Please see the attached example. The simulator allows to set the value of event E_1 only once, at startup. Then, the event value is not reset and cannot be changed. The edge with guard [valueof(E_1)==42] can only be traversed at startup (if the correct value of the event is set), and never again.

Am I doing something wrong, or is it an issue of the tool?

[image]
[image]

andreasmuelder, 2016-02-09:
This is a duplicate of #329

DominikTesch, 2016-02-09:
The view does not update itself.

arianaa30, 2016-02-18:
Is there an estimation of when it will be fixed?

andreasmuelder, 2016-02-22:
the bug is already fixed, there is only a minor update problem is the view. This will be rolled out in SCT 2.6.0.

andreasmuelder, 2016-02-24:
Fixed

closed 2016-02-24 by andreasmuelder