itemisCREATE/statecharts #291: C code generator should respect MISRA-C:2004 Rule 13.02 when using sc_boolean and related literals.

nyssen, 2016-01-11:
The type sc_boolean as well as the related literals (bool_true and bool_false) are used as a Boolean in all kinds of expressions. This leads to violations of MISRA-C Rule 13.02 ("Controlling expression is not an 'effectively Boolean' expression), which should be avoided.

That is, if a variable 'v1' of type sc_boolan is used in a Boolean expression, it should rather be 'converted' into a true Boolean expression by using '(v1 == bool_true)' instead. 
If the result 'r' of a  true Boolean expression is used as the value of an sc_boolean variable (or result type), it should rather be 'converted' by using a statement of the form 'r  ? bool_true : bool_false.

We should add these 'conversions' in all places where sc_boolean and Boolean are used in exchange.

BeckmaR, 2017-01-05:
What's the current state of this issue? Do the existing commits solve the issue or is further work needed? If yes, where?

terfloth, 2017-10-11:
is already implemented ....

closed 2017-10-11 by terfloth