itemisCREATE/statecharts #295: Usage of 'always' leads to superfluous code in case condition is also provided (C code generator)

nyssen, 2016-01-11:
When using 'always' on a transition together with a guard expression, the check function is generated with 'bool_true && ...' in its body, e.g.:

static sc_boolean sCT_check_Active_Calltr0_tr0(const SCT* handle) {
	return ((bool_true) && handle->ifacePhone.duration > 5) ? bool_true : bool_false;
}

This is superfluous and could be optimized away to only use the guard expression:

static sc_boolean sCT_check_Active_Calltr0_tr0(const SCT* handle) {
	return (handle->ifacePhone.duration > 5) ? bool_true : bool_false;
}

andreasmuelder, 2018-09-18:
@terfloth I think this is fixed with the current PR you are working on?

terfloth, 2018-11-21:
This issue still exists.

PR #2578 (2019-01-10): Issue 295
fixes #295
closed 2019-01-10 by terfloth