itemisCREATE/statecharts #300: Transition can specify multiple entry points

RainerKlute, 2016-01-13:
It is possible to specify multiple entry points in a transition, e.g. "# >failure >success". While it is reasonable to allow for multiple exit points, the number of entry points should not exceed one.

andreasmuelder, 2018-09-18:
multiple exit points should also be restricted

rherrmannr, 2018-09-19:
Multiple exit points are working in simulation. Also checked C code generation. There is a sctunit test, which uses exit states (ExitState.sct).

@RainerKlute with ``the number of entry points should not exceed one`` you mean the number of exit points used in the transition, correct?

rherrmannr, 2018-09-19:
Additionally, code completion does not work any more.

rherrmannr, 2018-09-20:
I've already implemented a validator for that issue. Meanwhile I have found the example statechart "UnboundEntryPoints02.sct" and modified it. 

Is this behavior correct? 

[image]

rherrmannr, 2018-09-20:
We need to define the behaviour for multiple entries.
Closed. See https://github.com/Yakindu/statecharts/issues/2273

PR #2272 (2018-09-25): Validation for multiple entry point spec
fix #300
commit ca0a75f2 (2018-09-25): Validation for multiple entry point spec

fix #300
closed 2018-09-25 by BeckmaR