itemisCREATE/statecharts #309: Couldn't resolve reference to EObject 'kv'

benedekh, 2016-01-17:
Dear All,  

I have an SCT model with an SGen file referred to it, and I would like to generate Java code from the statecharts. The only problem is, the reference in the SGen cannot be resolved to the respective SCT, even if I create a new SGen file and set the SCT reference in the wizard.  

I use YAKINDU Statecharts 2.5.0, and I attach the SCT and SGen files here: https://www.dropbox.com/sh/i898tbo7gp3nprw/AAAEs6hb7eUtK3nB-9DwX5T4a?dl=0  

May anyone know what went wrong?  Thank you for the help in advance.  

Regards, 
Ben

andreasmuelder, 2016-01-17:
this sounds as the Xtext nature is missing on your project.
Right click on the project that contains your sgen file and select configure -> add Xtext Nature. 
Does this work for you?

benedekh, 2016-01-17:
Unfortunately no, neither with YAKINDU SCT Nature.

andreasmuelder, 2016-01-18:
Closing this since it was resolved via the user group.
https://groups.google.com/forum/#!topic/yakindu-user/mdE2sqc29n8

closed 2016-01-18 by andreasmuelder