itemisCREATE/statecharts #315: Export statecharts as SCXML

developeron29, 2016-01-21:
Hi there,

Yakindu is such a great tool to create/edit statecharts. Thanks! 

I was working on a project that requires me to display/edit the statecharts that i create using yakindu on a web page. The best way to do that, that i could find was to export the statechart files as SCXML and displaying it using a SCXML javascript statecharts framework like jbeard4's SCION. But i was wondering if it was even possible to export the statecharts files from yakindu, in the scxml format. As the present .sct files seems to be formatted as basic xml. 

Any help would be deeply appreciated. Cheers.

terfloth, 2016-01-21:
@developeron29 : An SCXML export is not available currently. SCT makes use of XMI for storing the statechart. An SCXML export could be implemented as a code generator. Code generators are pluggable into SCT.

terfloth, 2016-02-22:
follow general interest in SCXML with Yakindu SCT 

see also: http://www-ens.iro.umontreal.ca/~syriani/courses/IFT6253-2016H/Project/topics.html

andreasmuelder, 2016-07-27:
This blog post may serve as a starting point, contributions are always welcome.
https://blogs.itemis.com/custom_code_generator_scxml

developeron29, 2016-07-27:
Thanks, I found an alternative to this,
Where i parse XML using DOM_Parser, fetch the required values and paste those values into a SCXML format.. Did the job, Thanks for all the help!

closed 2016-07-27 by developeron29

andreasmuelder, 2016-07-27:
@developeron29 nice, interested in sharing this with the community?

andreasmuelder, 2020-05-25:
just for documentation, YAKINDU has a SCXML code generator now.
https://www.itemis.com/en/yakindu/state-machine/documentation/user-guide/scxml_codegen