itemisCREATE/statecharts #325: Replace generator model typesystem with Generic Type System

andreasmuelder, 2016-01-25:
The generator model has a very simple type system implementation with 4 basic primitive types and basic type checks. This should be replaced by org.yakindu.base expressions to reduce duplicate code.

terfloth, 2016-02-22:
We have to check if this is topic for a major release as this may break compatibility.

andreasmuelder, 2017-09-07:
This would also allow to use expressions and variables in SGen for instance

GeneratorModel for yakindu::java {

     var targetSubFolder : string = "mySubFolder"
     var generateTimerServce : boolean = true
 
    statechart ECU_B {   

        feature Outlet {
            targetProject = "TargetProject"
            targetFolder = "target/" + targetSubFolder
            libraryTargetFolder = "src"
        }  
        
        feature GeneralFeatures {
            TimerService = generateTimerServce
        }
    }
}

closed 2017-09-12 by andreasmuelder