itemisCREATE/statecharts #326: C code generator

arelic, 2016-01-28:
Hello,

I have a question for you. 
I have installed the Eclipse development environment (for SCT v2.5.0) as indicated in your article. 
I have tried to make some minor changes in your C-code plugin, to avoid writing my own from scratch.

- Is there some way to export this plugin now to be regularly used in SCT environment?
- When I try to export it (org.yakindu.sct.generator.c project) I always get the message: "A cycle was detected when generating the classpath".  Is there any way to solve this or I am doing something completely wrong?

Best rergards,
Alex

andreasmuelder, 2016-01-29:
Hello Alex,
If you want to export your plugin without building it with maven you can select export / Deployable plugins and fragments from the context menu as you already did. To avoid problems with the existing c generator you should select another generator ID in the plugin.xml of your generator.c project.
```xml
 <SCTGenerator
            class="org.yakindu.sct.generator.c.CCodeGenerator"
            contentType="statechart"
            elementRefType="org.yakindu.sct.model.sgraph.Statechart"
            description="YAKINDU SCT C Code Generator"
            icon="icons/sct.png"
            id="yakindu::c"   <-- Change this
            name="YAKINDU SCT C Code Generator">
...
```
I tried to export the C Code generator this way and I also get the cycle error you mentioned. I think that this is an issue with Eclipse PDE - the message mentioned the plugin org.yakindu.sct.test.models in the cycle. This this is a test plugin and is not required by the production code so I am not sure why this is included. 
Anyway - just close this plugin before exporting, this should do the trick.
Let me know if this works!

What are the minor changes you implemented? Something that you want to contribute back to SCT?

Best,

Andreas

arelic, 2016-01-29:
Hello Andreas,
thanks for your quick reply. I'll try it out.

Actually the issue is that under certain circumstances, the c-code cannot be compiled due to inconsistent use of "const" specifier:

For example, this is how my function declaration is generated in **..._Required.h** file:

`extern sc_boolean gatt_server_profile_initIfaceInit_MyFunction(Gatt_server_profile_init* handle, const sc_integer profile_index, const sc_integer service_index, const sc_integer fsm_id, const sc_integer op_id);`

So, the _handle_ parameter **is not a constant**.

And then this function is called by passing it  a **constant** _handle_, which results with compilation error:

`static sc_boolean gatt_server_profile_init_check_main_region__choice_0_tr0_tr0(const Gatt_server_profile_init* handle) {
	return (bool_true) && bool_true == gatt_server_profile_initIfaceInit_MyFunction(handle, handle->internal.fsm_id, handle->internal.op_id);
}`

If I compile it, I get following error:

> Error[Pe167]: argument of type "struct <unnamed> const *" is incompatible with parameter of type "struct <unnamed> *" 

**Temporary Solution:**
What I did, just to be able to compile the code is to insert the const in declaration of my functions when generating **..._Required.h** file.
It solves this particular problem, but I am not sure if in some other constellation may cause some other side effects.

PS: I attached the .sct file which shall reproduce the c-code compilation problem

Best regards,
Alex
[gatt_server_profile_init.zip](https://github.com/Yakindu/statecharts/files/109949/gatt_server_profile_init.zip)

andreasmuelder, 2016-02-01:
Hello Alex,

thanks for the detailed bug report, seems that this is a regression introduced with version 2.5. We will fix this asap and provide a bugfix release.

Best regards,  
Andreas

nyssen, 2016-02-01:
Could you please attach your generator configuration (.sgen) as well?

arelic, 2016-02-02:
Here it goes :)

[ble_api.zip](https://github.com/Yakindu/statecharts/files/113592/ble_api.zip)

tkutz, 2016-02-26:
This should be now fixed with pull request #465

closed 2016-02-26 by tkutz