itemisCREATE/statecharts #328: Code generator must not execute if generator model is erroneous

RainerKlute, 2016-02-01:
The code generator must not execute if generator model is erroneous. If it does, a StackOverflowException is thrown:

`java.lang.StackOverflowError
	at org.eclipse.emf.ecore.resource.impl.URIMappingRegistryImpl$URIMapImpl.getURI(URIMappingRegistryImpl.java:162)
	at org.eclipse.emf.ecore.resource.impl.ExtensibleURIConverterImpl.normalize(ExtensibleURIConverterImpl.java:410)
	at org.eclipse.xtext.resource.XtextResourceSet$1.normalize(XtextResourceSet.java:290)
	at org.eclipse.emf.ecore.resource.impl.ExtensibleURIConverterImpl.normalize(ExtensibleURIConverterImpl.java:449)
	at org.eclipse.xtext.resource.XtextResourceSet$1.normalize(XtextResourceSet.java:290)
	at org.eclipse.emf.ecore.resource.impl.ExtensibleURIConverterImpl.normalize(ExtensibleURIConverterImpl.java:449)
	at org.eclipse.xtext.resource.XtextResourceSet$1.normalize(XtextResourceSet.java:290)`
…

andreasmuelder, 2016-02-02:
@RainerKlute I can't reproduce this can you attach an example please?

RainerKlute, 2016-02-08:
Sorry, @andreasmuelder, I don't remember exactly how I got there. I think the SCTUnit language hasn't been generated yet or something.

andreasmuelder, 2016-03-29:
reopen if it still exists

closed 2016-03-29 by andreasmuelder