itemisCREATE/statecharts #329: Simulation cannot be run as the value of an event can be set just the first time.

arelic, 2016-02-01:
Hello,

During simulation run the "value" of each event can be set just the first time until the event is the first time raised. 

After that the "value" cell in the table remains disabled and the new value for the same event cannot be entered any more.
[image]

Best regards,
Alex

andreasmuelder, 2016-02-07:
The type of the cell editor (string, real, integer, boolean) is selected by its java runtime type. When an event is raised, the value of the event isunset( = null) and the runtime type information is gone. This should be refactored that the type from the declaration is used to identify the right cell editor instead of the type of the runtime value.

andreasmuelder, 2016-02-08:
Fixed

closed 2016-02-08 by andreasmuelder