itemisCREATE/statecharts #330: Add error messages and solutions in user docu

SanniSch, 2016-02-01:
Mostly, the error messages in SCT show only the errors and not their solution. As the problems view has not the right format to integrate the solutions, it was decided to document the error messages in the user docu together with the solution and to link from the error messages in SCT to this parts of the docu.

Steps to do:

- List error messages (codes)
- Describe problems in user docu
- Add solution descriptions to the problems in the docu
- Link the error messages in SCT with the problems/solutions in the docu

terfloth, 2016-02-22:
We have to think about the introduction of error codes.

svenjawendler, 2017-05-02:
This can be connected to #1283, and #559.

SimonEit, 2017-08-22:
The scope of this seems more like an epic rather than a simple ticket. 

Suggestion: talk with @terfloth and @andreasmuelder about next steps

BeckmaR, 2017-11-10:
Closing in favor of #1742 where all error message issues are collected.

closed 2017-11-10 by BeckmaR