itemisCREATE/statecharts #333: Quickfix for adding "oncycle" or "always" to a transition

SanniSch, 2016-02-01:
If a transition has no name (no event) offer a quick fix to the user to add "oncycle" or "always".

SimonEit, 2017-08-22:
Suggestion: If a new transition between states is created by a user, do not create it with an empty transition label but pre-fill it with _always_

By doing this, new transitions don't create an immediate warning and just "work"

svenjawendler, 2018-02-15:
Now we have templates for this case. The user can add triggers by pressing ctrl+space.

Nevertheless quickfixes shall be added:
1. I add text to a transition -> quickfix: create an event for this text (questions: in which scope? in or out event?)

svenjawendler, 2018-02-26:
To solve this the sct infrastructure needs to be adapted to support quickfixes on stext models. The standard xtext solution https://www.eclipse.org/Xtext/documentation/310_eclipse_support.html#quick-fixes works on resources. This is not applicable for the statecharts. 

Though the quickfix solution for resources does not work with sct. Because SCTXtextIntegrationModule only implements Module and does not extend STextUiModule => it misses bindings.