itemisCREATE/statecharts #340: Add file extension automatically when user creates new statechart or code generator

SanniSch, 2016-02-01:
When creating a new statechart model or code generator, the user has to add the file extension (.sgen or .sct). However, the default title is highlighted completly: when the user starts entering text, the title and the file extension disappear.
-> If the user forgets to add the file extention behind the title, it should be automatically added when the statechart/code generator is created.

SimonEit, 2017-08-22:
Same issue was reported in #1280 

Current fix is that the error message shows the filetype.

@svenjawendler please close.

SimonEit, 2017-09-07:
[image]
Not the optimal solution for this issue for sure, but good enough for now. Usability Tests should be rerun on this, to see if this is actually an improvement for newbie-users.

closed 2017-09-07 by SimonEit