itemisCREATE/statecharts #360: State chart view comptetelly corrupted when object within state's region is moved.

arelic, 2016-02-03:
**Consequence:**
At the moment this constellation  (object with region inside) cannot be used at all, as it constantly corrupts the whole chart.

**How to reproduce:**
1) open attached .sct [tcu_message_handler.zip](https://github.com/Yakindu/statecharts/files/115890/tcu_message_handler.zip)
2) try to move the start entry node in "resend states r1 region "
3) the entire statechart gets totally corrupted.

**Before:**

[image]

**After moving the entry node (corrupted chart):**

[image]

Best regards,
Alex

andreasmuelder, 2016-02-03:
Alex,

it seems that the auto layout does not move the connection bendpoints.
I will check this issue, thanks for the detailed bug report!

Best regards,

Andreas

BeckmaR, 2016-10-07:
Just tested this, the issue is still there.