itemisCREATE/statecharts #364: In  generated C-code requests get lost

arelic, 2016-02-03:
Hello,

Please correct me if I am wrong, but in currently generated C-code, all events are cleared after a single _**RunCycle()**_ run although not all of them are served?

This practically means that in case you get more than one event at the time, all events except the one we used for transition from State-A  to State-B are forever lost (cleared).

**Example of real-life system:**
1) If you have one **server** (implemented in your state machine) and **5-clients** ,which can raise events to the server.
2) 3-clients raise the event to be serviced: event-1, event-2 & event-3.
3) Your state machine is currently in a state expecting one particular event-Z, which is by chance something else than above client events.
4) We call now the **_RunCycle()_** function, which sees that expected event-Z is not raised and therefore does nothing.
5) But before leaving the **_RunCycle() _**it **clears** all client events (event-1, event-2 & event-3). 
6) Events are now **lost **forever and clients (not knowing this) remain waiting forever to be served.

**Is there any recommended strategy how to overcome this situation?**

Best regards,
Alex

arelic, 2016-02-05:
I believe the same problem can be found is in Java code.
I guess this is a kind of conceptual issue.

**Could you please just shortly explain how to overcome above situation successfully?**

terfloth, 2016-02-05:
Hi Alex,

we are currently working on improving the documentation and this is definitely on of the topic.

First the behavior you describe is intended. It supports a basic execution model that can be used to implement different variants. It is important to understand the responsibilities of the generated code - so what the code cares about and what is out of scope. So the generated coded basically takes care about:

- providing the interfaces to access variables and events.
- initializing the state machine
- the execution semantics within a ‚run to completion‘ (RTC) step. This is everything within the runCycle function.

and there are some predefined constraints:

- the implementation is not thread safe
- runCycle must never be called in parallel
- all events for a RTC step (runCycle) will be consumed

Out of scope is:

- scheduling - the code generator can’t make any general assumption about scheduling mechanisms and constraints
- timers - the code generator does not know how the system handles times. So the behavior has to be contributed when required.
- event handling  - the code generator can’t know and does not prescribe how events enter the system. So deciding when events will be delivered to the state machine is the an outside responsibility.
- concurrency - as mentioned above the client code has to care about thread synchronization
- memory management - the generated statechart will never allocate any memory. Allocating the memory for the statechart structure is responsibility of the client code.

All these things are currently out of scope as we can’t provide a general solution without knowing more about the runtime system and libraries that are used. Better support for a more concrete usage context is often subject to code generator extensions.

We call the execution scheme that you applied (collecting events and then call runCycle) cycle-based. This allows to explicitly process several events at the same time and checking combinations of events (e.g. [eventA && eventB]). This is very useful for systems that are close to hardware and input signals. It basically is the IPO (input processing output) model. In your case you may better go with a event driven approach. You can implement that by directly calling runCycle when an event occurs so combine

   sc_raiseEventA(handle);
   sc_runCycle(handle);

Of course you likely have to care about buffering by using an event queue. As a queue implementation is out of scope for the reasons mentioned above this is the clients code responsibility. 

The Java code generator provides an option that generates a runnable wrapper. This runnable wrapper provides event queueing and multi threading support. This is easy to implement as a general solution for Java as  we simply rely on standard Java features. For C this is not as easy but as soon as it is defined what libraries to use the comparable thing can be provided by a code generator extension. This would save some manually written code. It may be worth to take a look at that Java generator feature - just add the RunnableWrapper feature to the seen configuration.

Best regard,
Axel  

Need professional support for Yakindu Statechart Tools?
Contact us: statecharts@itemis.de

arelic, 2016-02-09:
Hello Axel, 

thank you very much for clear and comprehensive answer.
I understand the background now.
 
I'll try to redesign my system to stay within the boundaries set by the generated code.

Thanks one more time.

Best regards,
Alex

andreasmuelder, 2016-02-19:
@RainerKlute can you add @terfloth s answer to the documentation

terfloth, 2016-02-22:
define location for documentation

terfloth, 2016-03-31:
place into 'general concepts of generated state machine code'

RainerKlute, 2016-03-31:
Fixed by pull request #540. @terfloth, please check and reopen, if needed.

closed 2016-03-31 by RainerKlute