itemisCREATE/statecharts #400: QtGenerator cannot be cast to org.yakindu.sct.generator.core.ISCTGenerator

stmork, 2016-02-14:
After setting up a complete custom generator infrastructure including update site I got in the destination project a QtGenerator cannot be cast to org.yakindu.sct.generator.core.ISCTGenerator. The QtGenerator is my Xtend entry point into the generator as generated by the new project wizard. 

After adding a Java class which extends the class GenericJavaBasedGenerator, implementing the runGenerator method and adjusting the extension point in the plugin.xml everything worked fine.

So I assume that this class should either generated in relation to the plugin.xml or the base classes of the Xtend generator entry class is not generated correctly.

andreasmuelder, 2016-02-19:
Thanks for reporting, we will check this if there is a problem with the generated infrastructure.

jdicks, 2016-03-01:
Hi Steffen,
we already fixed the class hierarchy for generated custom generators during #366. Additioanlly (see #473) a custom generator created by the wizard will now:
* add the CoreFeatureTypeLibrary features to get a proper configuration to use the Outlet feature out of the box when creating a generator model(sgen) for your custom generator
* register a file extension to get rid of a NoSuchElementException while using the generator model wizard
* last but not least the initial generators are now integrating with some changes done for #366 (basically the FileSystemAcces becomes more central, per default you don't need to handle IO by your own anymore, you can use FileSystemAccess.generate(fileName,content), the absolute targetfolder will be calculated by the FSA itself, standard is an EFS based FSA which e.g. refreshes resources implicitly after writing)

After changing these aspects it is now possible to export the generator created by the wizard by standard 'Export > Deployable Plug-ins & Fragments' and integrate it into an SCT distro out of the box.

All changes will be available in upcomming version, would be nice to get some feedback.

Regards, 
Johannes

closed 2016-03-01 by jdicks