itemisCREATE/statecharts #401: Document how to create a custom generator

stmork, 2016-02-14:
I needed about two days to setup a complete SCT custom generator infrastructure including maven build into an updatesite. So I assume that a lot of less experienced people have to spend more time to setup such a project. So it would be nice to have some more documentation how to develop a custom generator.

andreasmuelder, 2016-02-19:
How to build a custom generator with maven and create an update site is out of scope of YAKINDU Statechart Tools. The documentation on how to create a custom generator - especially more documentation about the different meta models - would be useful.

svenjawendler, 2017-05-02:
Is this still an issue? The documentation on this is 4 pages.

andreasmuelder, 2018-01-11:
This should be sufficient. Otherwise it is an issue for the professional support.

closed 2018-01-11 by andreasmuelder