itemisCREATE/statecharts #405: Consistency issue of "integer" type in Yakindu with "int" type of operation (custom java code)

arianaa30, 2016-02-16:
Apparently the type "integer" of Yakindu has some consistency issues with "int" of custom java code. BAsed on what I learnt, the statechart tools type 'integer' is mapped to the java type 'Long' for technical reasons. But this is not very intuitive. 

Maybe we should add some implicit type conversion from long to integer, at least for simulating operations. I had much problems figuring out what's going on!

terfloth, 2016-02-16:
The types used in SCT are named according to mathematical conventions like integral numbers and real numbers. From that point of view we use the scalar number type with the largest number range. And that is 'long (integer)' in Java. 

The Java code generator provides an option to enforce the use of int. But that is not possible for the simulator.

terfloth, 2016-02-16:
@arianaa30 : What is the context of your issue? Do you speak about the generated java code or simulation operation callbacks?

andreasmuelder, 2016-02-20:
@terfloth I think this is related to this discussion in our user group:
https://groups.google.com/forum/embed/?place=forum/yakindu-user#!topic/yakindu-user/7_u9yzA0GT8

terfloth, 2016-03-30:
Derive new issue...

closed 2016-05-10 by terfloth