itemisCREATE/statecharts #422: Place simulation view more prominently

SanniSch, 2016-02-18:
The simulation view is not prominent enough. Therefore there should be some improvements:
* Switch the position of the outline view and the simulation view
* The simulation view must be wider (currently only the a part of the first column is visible) -> the first and the second column should be visible
* The width of the columns should adapt automatically to the content

terfloth, 2016-02-22:
Derive tasks

svenjawendler, 2017-05-02:
@SanniSch  Is this ok for you?
[image]

terfloth, 2017-07-10:
* We should check if the simulation view layout is smart enough to adapt to view space. There should be no minimal size requirement.

SimonEit, 2017-08-22:
A follow-up suggestion for this issue:

During usability tests users reported that the current way the transitions are listed in the simulation view reminds them more of "hyperlinks" than actual buttons. Light blue underlined text appears as a link to an external source.

A smarter way would be to display the transitions as visual clickable buttons.

robert-rudi, 2017-11-14:
simulation view has been improved and new features are already merged.

closed 2017-11-14 by robert-rudi