itemisCREATE/statecharts #424: Improve user documentation

SanniSch, 2016-02-18:
There are some issues regarding the user docu summarized in this ticket:
* it is described that after creating a new Statechart a dialogue "Switch Perspective" appears. The dialogue does not appear -> delete from docu
* Meaning of keyword "internal" is not explained
* The user docu described that the context menu of src-folder has menu item "Generate Statechart Artifacts". In the tool it is "Generate Code Artifacts". It should be "Generate Code Artifact" in the docu.
* It is not clear in the docu where the simulation view is -> the name of the view is not completly visible in the screenshot. In addition, there should be borders with texts in the screenshot showing the areas belong to a view
* The chapter "Generating Java Source Code" is also a tutorial, however, this gets not clear because of the current structure. Put it to the tutorial chapters.
* There are several translation errors in the docu showing icons: e.g. "From the main menu, select File â†’ New â†’ Projectâ€Š: (Kapitel Getting startet > Creating an eclipse project" oder "Select General â†’ Project and click"
* Welcome dialogue: Topics: Statecharts: There is a blank after "(SCT" and bevor "..."
* Text is sometimes coloquial: “On the other hand your answer really doesn’t matter that much, because you can switch perspectives at any time.” or “Actually the new statechart isn’t really empty. It already contains the initial state (a small filled black circle), a “normal” state, and a transition leading from the former to the latter.” or “Hang on, we will correct the error in a second!” oder “Bingo! The state now has a proper name and the error marker disappears:”
* Create transition: "A light gray line is shown" -> is black with arrowhead in the tool
* Create a transition: picture does not fit to text ("Click on the On state") but screenshot shows that user has to click on "Off" state
* “Its background color changes to red while the On state’s background color becomes normal again.” -> off becomes normal again
* conclusion sentence at the end of the tutorial
* integrate a reference to the other two project examples
* the CallHandling example shows a state with spelling mistake (Dismiss Call)
* CallHandling Tutorial in Dismiss State "dura (new line)tion": there is a line break here that is wrong
* CallHandling: after creating SGen: Example differs from real situation

gjsjohnmurray, 2016-03-16:
It would be helpful for code copy/paste not to have the line numbers on the CallHandlingClient code, though I can see that you want to refer to the lines by number in the narrative text that comes after the code.

RainerKlute, 2016-03-17:
I'll update the following notes as this issue progresses:

> * it is described that after creating a new Statechart a dialogue "Switch Perspective" appears. The dialogue does not appear -> delete from docu

Probably you have already confirmed before that you don't want to see this dialog again. I'll add as hint in the documentation.

> * Meaning of keyword "internal" is not explained

To be clarified where *exactly* "internal" is not explained.

> * The user docu described that the context menu of src-folder has menu item "Generate Statechart Artifacts". In the tool it is "Generate Code Artifacts". It should be "Generate Code Artifact" in the docu.

To be verified.

> * It is not clear in the docu where the simulation view is -> the name of the view is not completly visible in the screenshot. In addition, there should be borders with texts in the screenshot showing the areas belong to a view

Hint to be added.

> * The chapter "Generating Java Source Code" is also a tutorial, however, this gets not clear because of the current structure. Put it to the tutorial chapters.

Well, that's our new documentation structure.

> * There are several translation errors in the docu showing icons: e.g. "From the main menu, select File â†’ New â†’ Projectâ€Š: (Kapitel Getting startet > Creating an eclipse project" oder "Select General â†’ Project and click"

Hm, I don't see what the problem is here, except you don't have proper '→' characters in your error message. In the documentation everything is looking good! Or am I missing anything here?

> * Welcome dialogue: Topics: Statecharts: There is a blank after "(SCT" and bevor "..."

To be verified.

> * Text is sometimes coloquial: “On the other hand your answer really doesn’t matter that much, because you can switch perspectives at any time.” or “Actually the new statechart isn’t really empty. [...]

Everything that adds to easier reading and more fun should be welcomed and appreciated, shouldn't it?

> * Create transition: "A light gray line is shown" -> is black with arrowhead in the tool

To be verified.

> * Create a transition: picture does not fit to text ("Click on the On state") but screenshot shows that user has to click on "Off" state

Fixed with issue #490.

> * “Its background color changes to red while the On state’s background color becomes normal again.” -> off becomes normal again

To be verified.

> * conclusion sentence at the end of the tutorial

To be considered.

> * integrate a reference to the other two project examples

What exactly do you mean? Where should references be added? And which project examples do you mean?

> * the CallHandling example shows a state with spelling mistake (Dismiss Call)

Fixed

> * CallHandling Tutorial in Dismiss State "dura (new line)tion": there is a line break here that is wrong

Verified; it's in the images. They should be redrawn.

> * CallHandling: after creating SGen: Example differs from real situation

What and where are these differences?

> * It is possible to add elements based on a graphical element (e.g. a state) by pressing Ctrl+Space. There is no information in the docu about this good feature -> should be added.

To be considered.

gjsjohnmurray, 2016-03-17:
Please see #491 for a screenshot showing how the '→' characters have been mangled in your Eclipse helpfile but are okay on your website.

closed 2016-03-23 by RainerKlute