itemisCREATE/statecharts #427: Builder should consider dirty state

SanniSch, 2016-02-19:
The builder does not consider the dirty state, this should be the case.
This leads e.g. to the following problem:
1. Create a new project
2. Create SCT model (do not save)
3. Generate SGen (do not assign it to a SCT)
-> The first state has a error that a state must have a name although it has one

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

BeckmaR, 2016-10-07:
Can't reproduce, possibly fixed already.

svenjawendler, 2017-05-02:
I cannot reproduce it. I assume it also as fixed.

closed 2017-05-02 by svenjawendler