itemisCREATE/statecharts #430: Shortcuts/Delete via keyboard does not always work

SanniSch, 2016-02-19:
In the statechart diagram it is no possible to delete elements via delete key. Also shortcuts such as Ctrl + Z do not work. This is a general problem on MAC (e.g. delete never works) but also happens sometimes on windows.

jdicks, 2016-03-31:
on windows shortcuts are working as expected, even if perspectives are changed. This might be an external issue. 
we still have to test this on mac.

tkutz, 2016-03-31:
My assumption is that the statement "Does never work on Mac" refers to using backspace which is [Fn+Backspace] on Mac or [Del] is extended keyboard is used.

tkutz, 2016-03-31:
@andreasmuelder @DominikTesch please test this and fill out the above table

jdicks, 2016-03-31:
@andreasmuelder observed that it becomes broken after editing via the properties view or editing wihtin specification items, but still not reproducible.

svenjawendler, 2017-05-02:
Windows 7: Ctrl+M, Ctrl+Y, Ctrl+Z, Ctrl+A, Ctrl+C, Ctrl+V, Ctrl+X work fine.

svenjawendler, 2017-05-02:
On Mac all the matching shortcuts also work.

closed 2017-05-02 by svenjawendler